thomas-haslwanter/scikit-kinematics #3: Support the scenario of optional magnetometer

tomfisher, 2016-11-13:
Hi, all the current algorithms in module imus require magnetometer data available there. It would be great if magnetometer is optional when the magnetic inference is a big issue in some applications.

Thanks.

thomas-haslwanter, 2016-11-13:
Hi Tom,
Thanks for the input.
All the systems currently implemented provide magnetometer data, therefore the "import_data" function always reads those in. As for the evaluation, the (exact) "quatInt" option ignores magnetomenter data. But all the published algorithms (implemented here) require magnetometer input. If you have a good algorithm that works without magnetomenter data, please let me know.
Also, contributions (in the form of implementations of other algorithms) would be greatly appreciated!

tomfisher, 2016-11-14:
"import_data" was one of the reasons that I did not start a pull request directly, which may affect your original design of the system. :) 

I know Madgwick implementations can work without the magnetic data, and don't know details about other algorithms. This is another reason that I did not start working on that.

thomas-haslwanter, 2016-11-14:
Can you send me the reference of Madgwick without magnetic field data? The paper "Estimation of IMU and MARG orientation using a gradient descent algorithm"  ( 2011 IEEE International Conference on Rehabilitation Robotics; Rehab Week Zurich, ETH Zurich Science City, Switzerland, June 29 - July 1, 2011, p 179-185) mentions explicitly "The MARG implementation incorporates magnetic distortion compensation."

thomas-haslwanter, 2016-11-14:
One more comment: on
http://www.olliw.eu/2013/imu-data-fusing/
I found the comment 
_"Madgwick has presented an interesting approach, which is based on formulating task T3 as a minimization problem and solving it with a gradient technique [SM]. I will argue here that this approach is – IMHO – not appropriate for IMUs which are using only gyro and accelerometer data (6DOF IMU)._ "

So any suggestion of which algorithms should best be used _without_ magnetometer info would be appreciated!

tomfisher, 2016-11-15:
Hi, man. I also read that excellent article.  :) The source code  quoted there should include what you need. 

I do agree 9DOF is more helpful to gain more accurate data. But 6DOF is more practical and less expensive than 9DOF.  In particular for some use cases in which the magnetic interference is a big issue, 6DOF is a better choice. 

Besides, 6DOF can achieve similar results as 9DOF can do if we turn parameters of filters carefully. A use case is PDR and can be found at [xio](https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU)