thomas-haslwanter/scikit-kinematics #8: ImportError: No module named 'imus'

kalenykkonstantyn, 2017-09-21:
Hello. 
If I run code from this example 
"# Set the in-file, initial sensor orientation
in_file = r'tests/data/data_xsens.txt'
initial_orientation = np.array([[1,0,0],
                                 [0,0,-1],
                                 [0,1,0]])
# Choose a sensor
from skinematics.sensors.xsens import XSens
..."
I get an error:
  File "/usr/local/lib/python3.5/dist-packages/skinematics/sensors/xsens.py", line 16, in <module>
    from imus import IMU_Base
ImportError: No module named 'imus'

Checked in Ubuntu and Win8 - same error

kalenykkonstantyn, 2017-09-21:
Thaks for Yours answer, but, unfortunatelly error is repeat:

The debugged program raised the exception unhandled ImportError
"No module named 'imus'"
File: /usr/local/lib/python3.5/dist-packages/scikit_kinematics-0.5.1-py3.5.egg/skinematics/sensors/xsens.py, Line: 16

thomas-haslwanter, 2017-09-25:
Hi Konstantyn,

I also went on my Unix system, but could not reproduce your error.
I am running

-          Mint 18.1

-          Python 3.6.0, installed with the distribution from Anaconda 4.4

-          scikit-kinematics 0.5.1

-          I ran "test_sensor_xsens.py", in the directory "skinematics\tests"
everything runs fine.

The things you can check:
When you start a terminal session, and type "pip show scikit-kinematics", do you get the correct information (0.5.1)?
When you open a terminal and start "python", do you get version 3.x?
When you type (in Python) "import skinematics", does it perform that command?
When you type "from skinematics import imus", does it perform that command?

If this does not work, we can also skype.

Best regards
th

thomas-haslwanter, 2017-09-25:
Hi Konstantyn,

I found out the problem: when trying to run the xsens file stand-alone, one dependency/path was incorrect (I only checked it run from other files).
If you update (now at 0.5.4), it should work.

Thank you very much for the feedback!

thomas

kalenykkonstantyn, 2017-09-26:
Hi Thomas! Thanks for fixing previous error, but began to appear this error:
tests/data/data_xsens.txt does not exist!
Exception "unhandled AttributeError"
'XSens' object has no attribute 'omega'
File: /usr/local/lib/python3.5/dist-packages/skinematics/imus.py, Line: 220

I changed the path to file, but this give no result

kalenykkonstantyn, 2017-09-26:
Source of example: http://work.thaslwanter.at/skinematics/html/imus.html#sub-classing-imu-base-for-your-own-sensor-type

thomas-haslwanter, 2017-09-26:
Hi Konstantyn,

sorry, but I cannot reproduce your error - neither on Windows, nor on Linux.

What I did:

I simply opened the file "imus.py" in its location (i.e. on my Unix system

~/anaconda3/lib/python3.6/site-packages/skinematics

and ran the program - and everything worked fine for me. One thing that makes me wonder, thought, is that on your system it is not under "site-packages", but in a folder "dist-packages". Don't you use a distribution like "anaconda" for your system?

I would be grateful if you could send me more information about your system, and about the error that gets generated, so I can try to locate the problem.

Best regards

thomas

thomas-haslwanter, 2017-12-18:
Hi Konstantyn,
I have made a few changes to the module "imus.py".
Could you please update, check if your problem still persists?
Thanks, thomas

thomas-haslwanter, 2019-04-06:
Since I believe the issue is solved, and since there have been no new comments for more than a year, I close that issue.

closed 2019-04-06 by thomas-haslwanter