thomas-haslwanter/scikit-kinematics #9: Data import error and unit for gyroscope data

jankodie, 2017-10-19:
Hi Thomas,

if I try to set sample rate, acceleration and gyroscope data by the `in_data` parameter of the `IMU_Base` constructor I get an error. The error is caused by the function call `self._set_info()` in line 199 of *imus.py* (last call in `_set_data` that is called with `in_data` in function  `__init__`). But the function expects all parameters in its definition (`def _set_info(self, rate, acc, omega, mag):` line 295 of *imus.py*). I solved it for me in the rather crude way to call it with 
`self._set_info(rate = self.rate, acc = self.acc, omega = self.omega, mag = self.mag)` from `_set_data` (line 199).

Besides that I have a question regarding the unit of the gyroscope data. In the comment for the `IMU_Base` class (line 65) the unit is mentioned as [deg/s] but in nearly all other cases as [rad/s] - as far as I've seen from a first look. I also didn't noticed a conversion between rad/s and deg/s. That's why I suppose the gyroscope data should be provided in [rad/s]. Is this correct?

Best regards,
dj

ivan866, 2017-12-17:
_set_info() in line 199 is called with no args, while it should have 4.

thomas-haslwanter, 2017-12-18:
Thanks for pointing that problem out: it crept in when I made the code more modular (for simpler implementation of new sensor), and forgot to update the manual data input.
Should be fixed in Ver 0.6.4.

closed 2017-12-18 by thomas-haslwanter

ivan866, 2017-12-22:
I provide `omega `in deg/s and `acc `in m/s^2 to `imus.in_data`, then call `imus.calc_position()` and `quat.quat2deg(imus.quat)`, and I get very large values in `imus.pos` (17000 to 1250000). What is wrong?

thomas-haslwanter, 2018-01-02:
1) In second order approximation,  `cumtrapz` is ok. However, you have to eliminate gravity first.
2) In order to get velocity, you should use scipy.signal.savgol_filter
3) IMUs measure the inertial signal with respect to the sensor. So the analysis has to be "bf"

iegorval, 2024-11-19:
> * In second order approximation,  `cumtrapz` is ok. However, you have to eliminate gravity first.

Do I understand correctly that you mean we should provide accelerometer data with already-removed gravity? Do you have any function doing that directly in scikit-kinematics?