thomas-haslwanter/scikit-kinematics #10: docs errors

ivan866, 2017-12-17:
http://work.thaslwanter.at/skinematics/html/view.html

>>> q = skin.quat.vel2quat(omegas, q0, rate, 'sf')

vel2quat seems to be deprecated, should be calc_quat

ivan866, 2017-12-17:
http://work.thaslwanter.at/skinematics/html/imus.html

imus.analytical argument order differs in some places, better be named arguments consistently

thomas-haslwanter, 2017-12-18:
Hi Ivan,
I fixed the first problem.
With the second I am not sure I understand you correctly: do you mean that the function "kalman" has the arguments in a different order (def kalman(rate, acc, omega, mag)) as analytical?
If that is the case, I have fixed it: now the argument sequence in analytical is:
def analytical(rate, accMeasured, omega, initialPosition, R_initialOrientation)

Please let me know if you had that in mind, or something else.

ivan866, 2017-12-18:
I meant that the docs on site and the source code have different argument orders, thus yielding an error if developer decides to use positional arguments rather than named.

The docs after your recent edits still don't match the sources, and you broke the interface in case someone used positional arguments.

thomas-haslwanter, 2017-12-20:
Is the current version ok? Or do you see any remaining problems?

ivan866, 2017-12-20:
Seems OK. You should also add `imus.pos` field. Pretty valuable.

thomas-haslwanter, 2017-12-21:
Done. Watch out, though, that minute offsets in the accelerometers can lead to erroneous values quickly!

ivan866, 2017-12-22:
`calculate_position=True` argument to `imus.IMU_Base` constructor is absent and thus effectively ignored in source 0.6.8

thomas-haslwanter, 2018-01-02:
`
    def __init__(self, in_file = None,
                 q_type='analytical', R_init = np.eye(3),
                 calculate_position=True, pos_init = np.zeros(3),
                 in_data = None ):`
This is the code for IMU_Base. Is it possible that you have an old version, and should update (currently 6.9)
And regarding the position calculation: I don't know your sensors. But my guess would be that the answer is a simple "no". For example, if you can accept an error of +/- 1 m, and you want to record for 10 min, then your accelerometer would need to have a drift of less than 6 micrometer/sec^2

thomas-haslwanter, 2018-01-22:
> http://work.thaslwanter.at/skinematics/html/imus.html
> “mahony” … Mahony algorithm, using, acc and omega
>
> In fact, Mahony uses mag too.

You are correct there – will be fixed in the docu.

thomas-haslwanter, 2019-04-06:
Documentation has been updated.

closed 2019-04-06 by thomas-haslwanter