thomas-haslwanter/scikit-kinematics #12: potential recursive import of 'quat'

patzm, 2017-12-18:
I freshly installed `scikit-kinematics`, version 0.6.3, with pip3 into a Python 3.4.3 installation on Ubuntu 14.04 x64 LTS. When I try to import `skinematics` or any module included, I get the following:
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/skinematics/__init__.py", line 29, in <module>
    importlib.import_module('.'+_m, package='skinematics')
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.4/dist-packages/skinematics/imus.py", line 31, in <module>
    from skinematics import quat, vector, misc, rotmat
  File "/usr/local/lib/python3.4/dist-packages/skinematics/quat.py", line 28, in <module>
    from skinematics import vector, rotmat
  File "/usr/local/lib/python3.4/dist-packages/skinematics/vector.py", line 24, in <module>
    from skinematics import quat
ImportError: cannot import name 'quat'
```
I think this is due to the fact that `vector.py` imports the `quat` module as well, which in turn is imported when the module is loaded.

thomas-haslwanter, 2017-12-18:
If two modules depend on each other, I don't see any other way. Also, if the module already exists in the current workspace, a new "import" should get ignored, unless it is forced.
On my Linux mint 18.1 system, I just uninstalled any previous versions, and re-installed scikit-kinematics. I was not able to reproduce your problem (the fact that I am running Python 3.6 should make no difference here).
Could you please update to 0.6.5, re-start the system, and try again?
Thanks, th

patzm, 2017-12-28:
I updated `scikit-kinematics` first to the most recent version 0.6.9 and then also tried with 0.6.5 as suggested. In both cases the error persisted. Here [I provide a list of other installed packages](https://gist.github.com/patzm/355f3d7e468457ce52d7065fe863003c). I hope that helps.

Furthermore I tried installing `scikit-kinematics` on a Ubuntu 17.10 x64 installation, which installs python 3.6.3 by default. Here it worked without problems.

Then I installed local environment with python 3.6.4 on the system that runs Ubuntu 14.04 LTS. Also in this local environment it worked... Did other users report issues with python 3.4 specifically?

thomas-haslwanter, 2018-01-02:
Hmm, I do remember that I did have some similar problems on an earlier version of Python. But I did not take any notes (and don't remember) which versions those were. Is there a specific reason why you don't upgrade to Python 3.6.x?
If it is a bigger problem for you, I will try to re-install an earlier version, and see if I can work around those installation specific problems. (Currently I don't have automated my tests, so that they automatically run for different versions of Python).

patzm, 2018-01-04:
Thanks for you offer but it is not an impediment atm. I did manually install python 3.6 in the end. I guess the default one that ships with Ubuntu 14.04 is python 3.4. That might be the only argument why to fix it. After all, a lot of docker images for deep learning are based off this distribution and I guess some people from this field use also the kinematics toolbox from scikit.

thomas-haslwanter, 2019-04-06:
Since there has been no new comment for more than a year, I close that issue.

closed 2019-04-06 by thomas-haslwanter