thomas-haslwanter/scikit-kinematics #14: view.orientation slow

ivan866, 2017-12-20:
view.orientation seems to cache the whole data when dealing with large datasets, thus yielding a long delay before showing the Tk window

thomas-haslwanter, 2017-12-20:
hmm, I have noticed that it is slow. But matplotlib seems to be a bit limited with animations, and using the 
`animation.FuncAnimation`
was the only way I thought I could do it.
Do you have any ideas how else to do it?

ivan866, 2017-12-20:
Probably should use OpenGL surfaces, anyhow this is how 3D plots are done in R.

thomas-haslwanter, 2017-12-21:
I have played around a bit with PyOpenGL, but still have a few problems. Do you have any experience with OpenGL?

ivan866, 2017-12-24:
Usually using OpenGL involves these steps:
getting GPU context
determining array of vertices
constructing them into triangles according to chosen winding rule
sending them to render

Pls check my comments in Closed issues #10, #9 section.

thomas-haslwanter, 2018-01-02:
Hi Ivan,

Maybe you can help me with this:

The attached program does a visualization with OpenGL (not the quaternions yet, but that is a small problem) - but I would really like to view the scene from higher up and right. Any idea what I would have to do?

Thank,

thomas

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

import pygame
import numpy as np
from pygame.locals import *

from OpenGL.GL import *
from OpenGL.GLU import *

import numpy as np
import pandas as pd
import abc

# To ensure that the relative path works
import os
import sys
dir_name = os.path.dirname(__file__)
sys.path.append(os.path.realpath(os.path.join(dir_name, "..")))

delta = 0.01
verticies = (
    (0, -0.2, -delta),
    (0, 0.2, -delta),
    (0.6, 0, -delta),
    (0, -0.2, delta),
    (0, 0.2, delta),
    (0.6, 0, delta),
    )

edges = (
    (0,1),
    (0,2),
    (0,3),
    (1,2),
    (1,4),
    (2,5),
    (3,4),
    (3,5),
    (4,5),
    )

colors = (
    (1,0,0),
    (0,1,0),
    (0,0,1),
    (0,1,0),
    (1,1,1),
    (0,1,1),
    (1,0,0),
    (0,1,0),
    (0,0,1),
    (1,0,0),
    (1,1,1),
    (0,1,1),
    )

surfaces = (
    (0,1,2),
    (3,4,5),
    (0,1,3,4),
    (1,4,2,5),
    (0,3,2,5)
    )

axes_endpts = np.array(
    [[-1,  0,  0],
     [ 1,  0,  0],
     [ 0, -1,  0],
     [ 0,  1,  0],
     [ 0,  0, -1],
     [ 0,  0,  1]])

axes = (
    (0,1),
    (2,3),
    (4,5)
    )

def draw_axes():
    glBegin(GL_LINES)
    glColor3fv(colors[4])
    #glLineWidth('+5')
    for line in axes:
        for vertex in line:
            glVertex3fv(axes_endpts[vertex])
    glEnd()
    
def Cube():
    glBegin(GL_TRIANGLES)
        
    for vertex in surfaces[1]:
        glColor3fv(colors[4])
        glVertex3fv(verticies[vertex])
        
    for vertex in surfaces[0]:
        glColor3fv(colors[0])
        glVertex3fv(verticies[vertex])
    #x = 0
    #for surface in surfaces:
        #for vertex in surface:
            #x+=1
            #glColor3fv(colors[np.mod(x,11)])
            #glVertex3fv(verticies[vertex])
            
    glEnd()

    glBegin(GL_LINES)
    glColor3fv(colors[2])
    for edge in edges:
        for vertex in edge:
            glVertex3fv(verticies[vertex])
    
    glEnd()
'''
'''

def main():
    #my_sensor = XSens(in_file=r'..\tests\data\data_xsens.txt')    
    
    pygame.init()
    display = (800,600)
    pygame.display.set_mode(display, DOUBLEBUF|OPENGL)

    gluPerspective(45, (display[0]/display[1]), 0.1, 50.0)
    glTranslatef(0.0,0.0, -5)

    angle = 0
    while True:
        for event in pygame.event.get():
            if event.type == pygame.QUIT:
                pygame.quit()
                quit()

        glLoadIdentity()
        angle += 1
        glClear(GL_COLOR_BUFFER_BIT|GL_DEPTH_BUFFER_BIT)
        glEnable(GL_DEPTH_TEST)
        glMatrixMode(GL_PROJECTION)
        glPushMatrix()
        glRotatef(angle, 1, 2, 1)
        Cube()
        glPopMatrix()
        draw_axes()
        pygame.display.flip()
        pygame.time.wait(10)

main()

ivan866, 2018-01-03:
The preliminary result is that by using `glRotatef` you rotate the camera, not the model. What you need to do is find a function which takes your model and rotates it like a 3D object in space, and just tilt your camera once beforehand.

ivan866, 2018-01-09:
Have tested `Orientation_OGL`. Works like charm.

closed 2019-04-05 by thomas-haslwanter