thomas-haslwanter/scikit-kinematics #15: Compensation for gravity hard-coded on 3rd axis

ivan866, 2017-12-23:
`imus.py`, line 358 - `g` is hard-coded in 3rd position, while MEMS accelerometer has vertical axis on Y, positive upwards.

`imus.py`, line 357 - `g` constant is hard-coded to 2nd digit, can be taken from `scipy.constants.g`

thomas-haslwanter, 2017-12-24:
I have changes line 357. 
Line 358 is correct, though. I always use a space-fixed coordinate system where the z-axis is defined by gravity. If the IMU is oriented differently, it has to be indicated (approximately) by adjusting "R_initialOrientation". 
The thing about coordinate systems is really confusing, and in fact, I am just about finishing a book on "3D Kinematics" for Springer, which is intended to provide the theoretical background when working in this area. What is your background, and how did you get into this area?

closed 2017-12-24 by thomas-haslwanter

ivan866, 2017-12-24:
I am now an affiliate researcher for some institute and we have a project which uses Tobii Glasses 2 eyetracker, which has MEMS gyro / accel built-in.

My question is how do I define an initial rotation matrix for a case when a person's gaze is headed towards Z axis, and vertical is Y positive, while left shoulder is X positive?

thomas-haslwanter, 2018-01-02:
import numpy as np

x = [0,1,0]

y = [0,0,1]

z = [1,0,0]

rot_mat = np.column_stack( (x,y,z) )

# assuming the data are in column form

rotated = (rot_mat @ data.T).T

should do the trick

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

ivan866, 2018-01-02:
That code simply reorders axes like that: X->Y, Y->Z, Z->X.

thomas-haslwanter, 2018-01-03:
... which is all that should be required if your coordinate system points in the wrong direction ....

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at