thomas-haslwanter/scikit-kinematics #17: Drift compensation

ivan866, 2017-12-24:
You used to have drift compensation in imus.py, line382, which equals 0.7*mean(acc). If I have mean=-10.02 m/s^2 in my data, does it mean I have an uncalibrated accelerometer and I need to use drift compensation then?

thomas-haslwanter, 2018-01-02:
No. There are two different points here:

1. Drift compensation would be acceptable, if I assume that i) everybody does the experiments in a static environment, and ii) the position at the end is (at least approximately) the initial position. I am somewhat reluctant to make that assumptions, especially the second one. Also, this is fairly easy to do on your own.
2. The second point relates to your offset. What you have is no drift, but the sensor aligned differently. You should rotate your data, by multiplying them with the corresponding rotation matrix.
I have just finished my book for Springer, in which I am explaining rotations, quaterions, analysis of IMU-data etc. Don't know when it will be published, but approximately in April. If you need it now, and you promise me that you will by the book when it appears, I could send you a pre-proof version.

ivan866, 2018-01-02:
Does it mean that because my MEMS 3-axis accelerometer was not oriented straight horizontally during the record, its vertical axis got a skew that does not equal usual `g` ?

thomas-haslwanter, 2018-01-02:
With a drift of 10 m/s^2, you are using a different coordinate convention. I use x forward, y left, z up. A misalignment would give you changes that are 2 orders of magnitude lower.

ivan866, 2018-01-02:
I've also noticed that the average X acceleration equals 1.05 and Z equals 0.67 even when the device was still! While Y is still -10.02 m/s^2.
My conventions are: +X to the left, +Y vertically up, +Z towards front.
And this type of accelerometer includes gravity component recorded in the data, so you need to filter it out first, and your package does, but, as I understand, because your program expects `g` to be recorded strictly as -9.81, but the data has another value, so the filtered data appears to be constantly accelerating, say falling down.

thomas-haslwanter, 2018-01-03:
Accelerometers that are that far off are a problem.

But before I implement anything, let's talk about it:

- Should I assume that there is exactly 0 movement at the beginning of the recording (first sample)?

- Should I assume that there is exactly 0 movement during the first 0.1 or 1.0 sec at the beginning of the recording?

- Should I make such a setting optional, or compulsory?

- I cannot make the assumption that there is such a strong drift, and that the device is exactly horizontal, can I? But that would also mean that I would interpret your drift primarily as a tilt of the device!

- Maybe it would be best to add a function that assumes that i) the device is exactly aligned with gravity, and ii ) that the drift is constant, and then stores those values in a calibration file. This could then optionally be called ....

But I think with values like that, you should contact the company, and ask for a working device ....

What do you think?

thomas

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

ivan866, 2018-01-03:
- Unfortunately, the records we've already made were not designed to have an obligatory still device at the beginning (the eyetracker was already put on the participant when the Start button was pressed).
- If we manage successfully to determine the source of such an error in this particular device, then. for sure, it should be recommended to hold the device still and strictly oriented for about 5-10 seconds from start.
- Optional, of course.
- I've also contacted the manufacturer's developers and asked them about the recommended way of drift compensation.
- Calibration file is the best thing I can think of for scientific research. But I'm not sure how to orient the device exactly horizontally. Should I use some sort of "liquid bubble level" tool, or a preinstalled table? The manufacturer should also guarantee that the accelerometer is well-oriented inside the body of the device.

thomas-haslwanter, 2018-01-03:
For your case, I would make the following assumptions:

- Assume that averaged over the whole experiment, the goggles don't slip, and gravity points downward.

- Assume that at the end, the subject is in approximately the same position as at the beginning.

With these two assumptions, you should be able to i) determine the orientation of the sensor with respect to gravity, and eliminate offsets. Then you should be able to track the movements during the recording.

It will probably be a bit of fuzzing around, but overall that should give you reasonable results.

BTW, which institute / company do you work at?

Best regards

th

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

ivan866, 2018-01-06:
Tobii (tobiipro.com) recommends to filter their data as described in this brochure: https://arxiv.org/pdf/1704.06053.pdf

I work for Russian Academy of Sciences (RAS) dept.

thomas-haslwanter, 2018-01-08:
Nice paper, thank you for the link!
Would be nice now if some would provide an implementation in Matlab / Python ;)

Best regards
th

ivan866, 2018-01-08:
@thomas-haslwanter, there you go: http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/

thomas-haslwanter, 2018-01-09:
Those algorithms are already included in skinematics.

I have now finished adding the OpenGL-viewer, and update scikit-kinematics to Ver 0.7.0

Best regards

th

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

ivan866, 2018-01-09:
Have tested `Orientation_OGL`. Works like charm.

ivan866, 2018-01-09:
> ...subject is in approximately the same position...
How approximate is approximately? 20 cm, 1 m, 5 mm?

thomas-haslwanter, 2018-01-10:
I would say about 0.5 m

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

thomas-haslwanter, 2019-04-06:
No open questions here, and no more comment for more than a year. So I close the issue.

closed 2019-04-06 by thomas-haslwanter