thomas-haslwanter/scikit-kinematics #18: Test data

ivan866, 2018-01-20:
madgwick filter somehow rotates the data 90 degress around X axis:

```
from skinematics.sensors.xsens import XSens
from skinematics.view import Orientation_OGL

in_file = r'd:\myProject\venv\Lib\site-packages\skinematics\tests\data\data_xsens.txt'

sensor = XSens(in_file=in_file, q_type='madgwick')

viewer=Orientation_OGL(quat_in=sensor.quat)
viewer.run()
```

ivan866, 2018-01-20:
What are the conversion factors in `polulu.py`? What are they for?

thomas-haslwanter, 2018-01-22:
Factors to convert the digital numbers to SI-values. I have taken them from a Swiss student of mine, and not checked any documentation. It seems to be a cheaper sensor.

Best regards
th

thomas-haslwanter, 2018-01-25:
In that experiment the sensor was strapped to the side of the head, and thus rotated by 90 deg. So this is not due to the Madgwick filter, but due to the paradigm.

---
Prof. (FH) PD Dr. Thomas Haslwanter
School of Applied Health and Social Sciences University of Applied Sciences Upper Austria FH OÖ Studienbetriebs GmbH
Garnisonstraße 21
4020 Linz/Austria
Tel.: +43 (0)5 0804 -52170
Fax: +43 (0)5 0804 -52171
E-Mail: Thomas.Haslwanter@fh-linz.at
Web: bim.fh-linz.at
or work.thaslwanter.at

thomas-haslwanter, 2018-01-25:
In that experiment the sensor was strapped to the side of the head, and thus rotated by 90 deg. So this is not due to the Madgwick filter, but due to the paradigm.

closed 2018-01-27 by thomas-haslwanter