thomas-haslwanter/scikit-kinematics #26: Sample rate vs. Sample period

Massendefekt, 2019-01-02:
There seems to be an inconsistency with the use of the sample rate when comparing the Kalman filter to the Madgwick filter. 

The Kalman filter takes a rate in Hz as input (which is later converted into seconds) while the Madgwick filter takes a sample period in seconds. This makes it inconvenient to use the same `input_data` for both filters.

Is it possible to adjust this issue? I'm also ready to submit a PR if needed.

lembit1, 2019-02-21:
I strongly support standardizing on using either sample rate or period. In some places the rate is integer type, while really should be able to use float.

thomas-haslwanter, 2019-03-20:
I agree with the suggestions. They are implemented in the next release.

closed 2019-03-20 by thomas-haslwanter