thomas-haslwanter/scikit-kinematics #27: Madgwick and Mahony 6Dof

alexx600, 2019-01-26:
the original Madgwick and Mahony filter are available in 6DoF and 9DoF versions (http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/). It would be great if the project enabled the use of 6DoF as well. In fact, I need them in such versions.

mathiasverbeke, 2019-03-07:
That would indeed be very interesting. I'm also looking for a 6DoF version of these algorithm. Or is there a way to use the 9DoF versions with 6DoF data?

alexx600, 2019-03-09:
> That would indeed be very interesting. I'm also looking for a 6DoF version of these algorithm. Or is there a way to use the 9DoF versions with 6DoF data?

Algorithms in version 9 of the DoF must have additional information about the intensity of the magical field. You can not put "1" or "0" there.

Theoretically, such indications can be generated eg using the IMUsim project. However, there is no logic in such a thing.

In link i mentioned there is open source class to MATLAB software.