thomas-haslwanter/scikit-kinematics #31: the Kalman implementation questions

stweiyy, 2019-02-22:
according to my understanding, the implementation of the Kalman seems have 2 problems:
(1)
...
z_k_pre = z_k.copy()
... 
# Update state vector x_k
x_k += np.array( K_k.dot(z_k-z_k_pre) ).ravel()
here z_k - z_k_pre means the current observation subtract the previous observation. according to standard Kalman text book, shouldn't here be observation subtract prediction ? and why the prediction part is just simply set to the previous observation (z_k_pre = z_k.copy())
(2)
....
x_k[:3] = np.zeros(3)
x_k[:3] += tstep * (-x_k[:3]+z_k[:3])
....
here the first 3 elements is forced to 0. so actually it is 
x_k[:3] += tstep * (z_k[:3]), 
since z_k[:3] is the angular velocity,  the result is the amount of changed angles.  
which does not make sense.

thomas-haslwanter, 2019-03-21:
I am just looking into that again, and I am seriously wondering if my implementation was correct. Currently, I cannot even find the definition of the QUEST that I used at the time :(
Sorry - Will keep you updated.

gcosne, 2020-07-20:
Hi @thomas-haslwanter, 

I can see that the Kalman code have been changed since @stweiyy comment. Do you have any update on the definition of QUEST you used ?
How confident are you in the Kalman implementation? 

thank you for your help