thomas-haslwanter/scikit-kinematics #33: ImportError: cannot import name 'dist' from 'matplotlib.mlab'

divan, 2019-06-25:
Hi,

I installed scikit-kinematics on my MacOS X system via pip3 (`pip3 install scikit-kinematics`), installation went smoothly, but when I try to import library it throws an error:

```
$ python3
Python 3.7.2 (default, Feb 12 2019, 08:15:36)
[Clang 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import skinematics as skin
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/skinematics/__init__.py", line 29, in <module>
    importlib.import_module('.'+_m, package='skinematics')
  File "/usr/local/Cellar/python/3.7.2_2/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.7/site-packages/skinematics/view.py", line 34, in <module>
    from matplotlib.mlab import dist
ImportError: cannot import name 'dist' from 'matplotlib.mlab' (/usr/local/lib/python3.7/site-packages/matplotlib/mlab.py)
>>>
```

Any suggestion what can be wrong with my setup?
Thanks in advance.

vinay-k-pisharody, 2019-06-29:
I am getting the same error with Python 3.6 as well.

jeffreymonaco, 2019-08-02:
I encounter this too.

The issue appears to be that scikit-kinematics 0.8.1 is using functions that have been removed from matplotlib as of matplotlib 3.1.0.

Specifically, matplotlib documentation indicates that the function mlab.dist was deprecated in matplotlib 2.2, and it has been removed in matplotlib 3.1.0. The suggestion is to use the numpy function hypot.  See https://matplotlib.org/3.1.0/api/api_changes.html for a discussion of this API change.

I think the ideal fix is for scikit-kinematics to be updated for compatibility with the current matplotlib version. Absent this, you can set up a custom python environment using an earlier version of matplotlib. As a last resort, you can probably edit the scikit-kinematics offending file (view.py) to use numpy.hypot instead of mlab.dist.

I just encountered this problem so I haven't gone through any of the above steps to know if there are similar gotchas.

Good luck.

thomas-haslwanter, 2019-08-05:
Should be fixed now. 
Sorry for the slow implementation - I have been sick :(

closed 2019-08-05 by thomas-haslwanter

Jendker, 2019-09-04:
Is there any chance of uploading the new version to pip?

marcusnh, 2021-09-06:
Is this problem fixed? I have encountered the same problem when installing  with pip

dadasmash, 2024-04-15:
here's a solution that manully define it: https://github.com/ESMG/pyroms/issues/18