thomas-haslwanter/scikit-kinematics #35: Enable scikit-kinematics to run without OpenGL

biernackip, 2019-11-19:
It would be tremendously helpful if the library could be used only for the calculations, i.e. without OpenGL.

Currently, when we want to use `from skinematics.imus import IMU_Base`, then the `__init__.py` is called, which triggers the import of `view` (which in turn imports OpenGL)

Perhaps there could be an if statement that checks if OpenGL is installed.

Please let me know if you would like me to provide a solution in the form of a PR

thomas-haslwanter, 2020-07-01:
Sorry about the tremendous delay: I had health issues, which have prevented me from attending to scikit-kinematics.
A PR would be very much appreciated.
Also, without OpenGL animated 3D visualizations don't work well. Why do you want to skip it?

thomas

yustiks, 2020-10-21:
I also encountered problems with GPU as my laptop doesn't want to run the library without it

yustiks, 2020-10-21:
> Sorry about the tremendous delay: I had health issues, which have prevented me from attending to scikit-kinematics.
> A PR would be very much appreciated.
> Also, without OpenGL animated 3D visualizations don't work well. Why do you want to skip it?
> 
> thomas

Oh, sorry to read it, hopefully, you are better now!