thomas-haslwanter/scikit-kinematics #37: Real time application

shantnavagarwal, 2020-02-07:
Is it possible to use this library for real-time analysis? According to the documentation, the library requires recorded data. I want to use this library while streaming data from my IMU (BNO055 usb stick)

thomas-haslwanter, 2020-02-11:
The toolbox is primarily intended for off-line analysis. And to be fully portable, it is kept in pure Python. One can speed it up massively - without much work - by using e.g. Cython. But then you get a platform-dependent result, and I did not want to go that way.
It may work, depending on your task. For computationally more challenging tasks, such as converting raw-data to quaternions, it will probably depend on your sampling rate. So my suggested steps would be:
i) make sure that you have the minimum sample rate that is sufficient for your task
ii) then give it a try with the existing toolbox
iii) if that does not work, speed up the slow parts of the code with Cython
iv) and as a last resort, implement the algorithms in C.

If you get to (iii) and have difficulty, let me know.
thomas

closed 2020-02-12 by thomas-haslwanter

shantnavagarwal, 2020-02-13:
This sounds like a plan!
1) sampling rate is 400hz
2) I am working on this now

I will let you know about my progress. Thanks