thomas-haslwanter/scikit-kinematics #44: Xens data file doesn't work

cab938, 2021-04-21:
I've got binary datafiles from Xens (.mvn) but it seems the code is setup to work with a different kind of file (a .csv?). Maybe the format has changed, or maybe I need to preprocess it first somehow?

mustafa-sarshar, 2021-05-17:
no sir, actually it is compatible only with .txt file with skiprows=4 and sep=","
If you check the codes, you will find it very straight forward.

misz11, 2023-05-26:
I have an Xsens IMU and would like to use the package skinematics.
However, the Xsens class of yours says the data is generated by:
Import data saved with XSens-sensors, through subclassing "IMU_Base".

Where can I find this data generation. With the SDK from the manufacturer, a txt.file is created but not in the desired format of your Xsens class, and therefore does not work.
If this doesn't exist, it doesn't make sense from my point of view to make different classes of sensors, if the data is read from txt files anyway and there is no possibility to read the sensors in the right format.

Sorry for my confusion.

thomas-haslwanter, 2023-05-26:
I am not sure if I understand your question.
If your (new-format) data are in a single file, then

- copy `skinematics/sensors/xsens.py` to a file `xsens_23.py`
- open this file, and modify the method `get_data` (lines 26-67) so that they work with the new file
- rename (in `xsens_23.py` the class `XSens` to `XSens_23`

and you should be able to use the new module.
Ideally, if it works you would make this a pull request.

Note that you have to have this file in the module where skinematics is found. You can find this by opening Python, and type
`import skinematics as skin
 skin.__file__`

misz11, 2023-05-26:
And how did you record the data from the Xsens IMU?
perhaps my implementation of the data recording is faulty.

thomas-haslwanter, 2023-05-26:
Can you post the first 10 lines of your data-file?

misz11, 2023-05-31:
Acc X: 0.04, Acc Y: -0.03, Acc Z: 7.35 |Gyr X: 0.00, Gyr Y: -0.01, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.06, Acc Y: -0.03, Acc Z: 9.79 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.03, Acc Z: 9.79 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.02, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.04, Acc Y: -0.03, Acc Z: 9.81 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.04, Acc Y: -0.03, Acc Z: 9.81 |Gyr X: 0.01, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.04, Acc Y: -0.04, Acc Z: 9.79 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.03, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.03, Acc Z: 9.81 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.04, Acc Y: -0.04, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.04, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.03, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.01, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.05, Acc Y: -0.03, Acc Z: 9.79 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24
Acc X: 0.04, Acc Y: -0.03, Acc Z: 9.80 |Gyr X: 0.00, Gyr Y: -0.00, Gyr Z: 0.00 |Mag X: 0.31, Mag Y: 2.29, Mag Z: 0.24

this are my first 10 lines

thomas-haslwanter, 2023-06-02:
This looks weird. Is this really the file produce by the XSens recording software? This looks more like data that have already been read in by you!

A Python script to read in the latest version of XSens-sensors can be found [here](https://base.xsens.com/s/article/Reading-an-MVNX-file-with-Python?language=en_US).
Please try to figure out (from your XSens-documentation) what format your data-file is, and proceed from there.

misz11, 2023-06-02:
This is the file created by the example python code of the SDK (provided by Xsens/Movella). The data is first read and stored in a log file and with another script this txt file can be created from this log file as shown above.

I have not changed anything in the code.